fossasia/pslab-app #934: Show a form under Collect Issues

cweitat, 2018-05-28:
**Actual Behaviour**
Currently is redirecting to PSLab's issues on Github

**Expected Behaviour**
Create a form to collect user input and collate elsewhere like Google Form instead of showing the Github pages. 

**Screenshots of the issue**
[image]
[image]

**Would you like to work on the issue?**
@abhinavraj23

abhinavraj23, 2018-05-30:
@cweitat @CloudyPadmal I make the google form, I have also created the temporary Google account of PSLab for that, but for submitting issues we also need a personal access token of a Github account.

cweitat, 2018-05-30:
Don't think we will be submitting issues automatically. Someone will look through the submitted responses before opening an issue in case there's duplicate

abhinavraj23, 2018-05-30:
So how do we plan to do that? I mean don't you want that the submitted responses automatically goes to the Github issue or do you want to manually open a issue after looking through the responses?

CloudyPadmal, 2018-05-30:
@cweitat we have to do it manually. @abhinavraj23 just the form collection is enough. We can review them and add the relevant issues to tracker manually. Do it fast!

abhinavraj23, 2018-05-30:
Sure @CloudyPadmal :+1:

abhinavraj23, 2018-05-30:
@cweitat @CloudyPadmal Shall I share the Google account details here?

abhinavraj23, 2018-05-30:
@CloudyPadmal @cweitat Please suggest me the format of the Google form, I thing it can be of type:-
          1) Issue title.
          2) Description about issue.
          3) Email address (of the user)
What do you guys think?

cweitat, 2018-05-30:
Is it possible for user to upload image to show screenshot through form? It can be just like the template of the issue excluding the assigning to who and logs part

harsh-2711, 2018-05-30:
@abhinavraj23 I think there's no need to add email ID column as google itself identifies the current email ID being used on the device and also gives option to switch the account in the form itself while sending the form.

abhinavraj23, 2018-05-30:
Also,now we have two methods to implement this form in the App, either we directly go for the web-view or we create a new UI in the report issue section in which we can we will have questions and area for answer (same as in google form) and we can do the coding for getting the responses through the app. Do let me know which is better. :+1:

abhinavraj23, 2018-05-30:
@harsh-2711 I have not put that email-address columns, it just came by itself when I enabled the get-email feature of Google forms. I am also confused in this part.

PR #942 (2018-05-31): feat: added Google Form for report issue
Fixes #934 

Changes: Added the Google form for issue collection.

Screenshots for the change: 
[image]

APK for testing: 
[app-debug.apk.zip](https://github.com/fossasia/pslab-android/files/2054958/app-debug.apk.zip)
closed 2018-05-31 by CloudyPadmal