fossasia/pslab-app #953: Implement PV sources in Power Source Instrument

CloudyPadmal, 2018-06-02:
**Actual Behaviour**

Current Power Source instrument is just boilerplate code.

**Expected Behaviour**

Programmable Voltage Sources need to be implemented and tested.

**Steps to reproduce it**

1. Open app
2. Go to `Power Sources`

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Yes

abhinavraj23, 2018-06-02:
@CloudyPadmal I have completed my milestone task, so now if you want then both of us can work on the implementation of power source.

CloudyPadmal, 2018-06-03:
@abhinavraj23 Let's finish one instrument fully and then move onto the other one. Now that you have completed the UI, you can open up new issues and start working on coding implementation and test it with the device.

PR #995 (2018-06-11): fix: implement functionality in power source instrument
Fixes #953 

Changes: 
- Added controllability of power sources using Android app

TODO:
- Long click on increment decrement buttons to update values repeatedly #996 
- Dialog box to show information #997
- Make changes to instrument tile #998
- Add device connectivity menu buttons #999
- Save user set values throughout the app #1000

Screenshots for the change: 

| Device | Multimeter |
| -------- | -------------- |
| [image] | [image] |

APK for testing: 
[PowerSources.apk.zip](https://github.com/fossasia/pslab-android/files/2084397/Pow [... cut]
closed 2018-06-11 by CloudyPadmal