fossasia/pslab-app #969: Unhandled threads in sensors fragment

harsh-2711, 2018-06-04:
**Actual Behaviour**

When any sensor is opened from the given list, then on pressing back button crashes the app

**Expected Behaviour**

The app shouldn't crash, rather main fragment main fragment of sensor activity should pop up

**Steps to reproduce it**

Go to sensors and use any sensor from the list and then press back key

**LogCat for the issue**

Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

harsh-2711, 2018-06-05:
@CloudyPadmal  Can I convert the sensors fragment to there respective activity as it will solve this and the subsequent issues that are opened by me. It will be easier to handle threads and layouts for activity rather than overlapping fragments. Also the sensordock used is always cut out because the layouts aren't designed properly. So should I proceed with this idea or do you have any other better solution?

CloudyPadmal, 2018-06-05:
Yes that would be better! :+1:

PR #987 (2018-06-15): fix: Changed fragments to activities for evey sensors
Fixes #969 #970 #984 #986 

Changes: Changed fragments to activities for every sensor and removed the placeholder fragment from main sensor activity and updated the sensor activity to show the sensors list solving orientation issue. Also provided suitable margins to the sensor dock and the layout to provide better UI. Also the issue of orientation is handled by creating activities.

Screenshots for the change: 
[image]

APK for testing: [app-debug.zip](https://github.com/fossasia/pslab-android/files/2078495/app-debug.zip)
closed 2018-06-15 by CloudyPadmal