fossasia/pslab-app #972: Implement 'How to use' section in all the instruments

yatri1609, 2018-06-05:
**Actual Behaviour**
Right now there are no directions on how to use a particular instrument or a document helping new users implemented in the app.

**Expected Behaviour**
We must have one to help new users get acquainted to the app by themselves or can use it in some confusion.

**Steps to reproduce it**
Open app and open any instrument

**LogCat for the issue**
Not applicable

**Screenshots of the issue**
Not required

**Would you like to work on the issue?**
Yes I would like to implement a help button on the top which can be clicked to refer to the 'How to use' guide.

yatri1609, 2018-06-05:
@CloudyPadmal Should I implement it and continue with this issue?

CloudyPadmal, 2018-06-05:
Yes please! :+1:

yatri1609, 2018-06-06:
@CloudyPadmal All the instruments are not having the common layout as the title bar is missing in a few like that of Multimeter and Oscilloscope so rather than for each instrument can I implement a common guide for all in the main Instrument section itself ?

yatri1609, 2018-06-06:
@CloudyPadmal Or we can implement the dialogue box in each instrument. Which one will be more effective?

yatri1609, 2018-06-08:
I have implemented this. Suggest the changes except that of color scheme as I will be doing that later today.

[image]
[image]

@CloudyPadmal  Please have a look at this and suggest me the changes and whether I should make a PR for this or not.

yatri1609, 2018-06-08:
@Vikum94 @harsh-2711 @abhinavraj23 @Avjeet Please have a look at the same and provide suggestions.

CloudyPadmal, 2018-06-08:
@yatri1609 this is nice! :+1: can you get started?

yatri1609, 2018-06-08:
Ya I'll implement it for all the buttons.

yatri1609, 2018-06-14:
Since I'm not finding the in-boarding guide as per our requirement. @CloudyPadmal Should I implement the same with the dialog box?

yatri1609, 2018-06-14:
[image]

Something like this? I think we can do this for the more important features explaining the users what is it and what does it shows.

CloudyPadmal, 2018-06-14:
@yatri1609 yes then we can go for an implementation like this! :+1: You can get started!