fossasia/pslab-app #976: Multimeter shows various values for a single resistance

abhinavraj23, 2018-06-06:
**Actual Behaviour**

Multimeter shows various values for a single resistance  when clicked multiple time(as every resistance has some tolerance level)

**Expected Behaviour**

It should show a single value which should be average of 10 values taken.

**Steps to reproduce it**

Run an iterative loop 10 times and calculate the average.

**LogCat for the issue**

Not Applicable

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-06-06:
@abhinavraj23 what if user changes the resistor during the loop?

abhinavraj23, 2018-06-06:
@CloudyPadmal I was thinking about that, but I have observed that in 10 observations the average becomes constant and thus we have to limit the loop till 10, which according to me will not take that time so as the user changes the resistor, what do you think about it, we can also keep a toast telling the user to keep the resister intact for few minutes. Please comment

PR #990 (2018-06-08): refactor: reduced fluctuation of resistance in multimeter
Fixes #976 

Changes:added a loop which calculates the average resistance taking 20 values, thus gives a stable value

Screenshots for the change: 
[image]

APK for testing: [Compress the *.apk file into a rar or zip file and upload it here]
[app-debug.apk.zip](https://github.com/fossasia/pslab-android/files/2080492/app-debug.apk.zip)
closed 2018-06-08 by CloudyPadmal