fossasia/pslab-app #998: Improvements on Power Source Tile

CloudyPadmal, 2018-06-08:
**Actual Behaviour**

Tile image for power source looks big and mis-aligned

**Expected Behaviour**

Tile image needs to be aligned properly with correct margins around them, even in portrait mode.

**Steps to reproduce it**

1. Open app
2. Scroll down instruments list for `Power Source`

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

cweitat, 2018-06-12:
Multimeter and logical analyzer does not looks good as well. 

[image]

This problem affects based on devices' resolution. On high res devices, the some icons look normal.

CloudyPadmal, 2018-06-13:
Yes @cweitat ! The reason is the whole card is made as an image. So it doesn't stretch out properly in different screens. Let me have a fix for this one! :+1:

PR #1048 (2018-06-16): fix: uniform tile icons design update
Fixes #1024 
Fixes #1025 
Fixes #1026  
Fixes #1027 
Fixes #1028  
Fixes #998  

**Changes**: 
Designed tile icons to have a common dimensions and theme as requested in the weekly meeting.

**Screenshot/s for the changes**: 

|  |  |  |
| ----- | ----- | ----- | 
| [image] | [image] | [image] |

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [X] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [X] No modifications done at the end of resource files `string [... cut]
closed 2018-06-16 by CloudyPadmal