fossasia/pslab-app #999: Add device connection menu items to power source

CloudyPadmal, 2018-06-08:
**Actual Behaviour**

In the top action bar, device connectivity menu items are not made available

**Expected Behaviour**

Similar to `Instruments` page, top action bar requires the menu items in `Power Source`

**Steps to reproduce it**

1. Open app
2. Go to `Power Source` and observe the action bar

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

cweitat, 2018-07-15:
@CloudyPadmal currently the compass on the instruments page has nothing yet. Can we have a toast when a user press on it to show text like "Still in development. Please check next time."? Otherwise user may think that it is the app/phone not working. Thanks

CloudyPadmal, 2018-07-15:
Yes @cweitat ! I'll make that change and send a PR in a while! :+1:

neel1998, 2019-03-29:
@CloudyPadmal  @mariobehling  @cweitat  The compass instruments screen has now been implemented. So there is no need for the toast. I think this issue can be closed

closed 2019-04-29 by cweitat