fossasia/pslab-app #1011: Lux Meter crashes the app when the fragment is destroyed

harsh-2711, 2018-06-11:
**Actual Behaviour**

On closing Lux meter instrument crashes the app

**Expected Behaviour**

The app shouldn't crash and the instruments page should open

**Steps to reproduce it**

Open Lux meter from Instruments section and then close it by pressing back key

**LogCat for the issue**

E/AndroidRuntime: FATAL EXCEPTION: main
                  Process: org.fossasia.pslab, PID: 14627
                  java.lang.IllegalStateException: Fragment LuxMeterFragmentData{5360359} not attached to a context.
                      at android.support.v4.app.Fragment.requireContext(Fragment.java:614)
                      at android.support.v4.app.Fragment.getResources(Fragment.java:678)
                      at android.support.v4.app.Fragment.getString(Fragment.java:700)
                      at org.fossasia.pslab.fragment.LuxMeterFragmentData$SensorDataFetch.onSensorChanged(LuxMeterFragmentData.java:182)
                      at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:865)
                      at android.os.MessageQueue.nativePollOnce(Native Method)
                      at android.os.MessageQueue.next(MessageQueue.java:325)
                      at android.os.Looper.loop(Looper.java:142)
                      at android.app.ActivityThread.main(ActivityThread.java:6753)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:482)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-06-11:
@harsh-2711 how's the progress with Logic Analyzer functionalities? Have you been able to fetch data from PSLab? Try to focus on the assign instrument and make it complete first before moving on to other instruments.

harsh-2711, 2018-06-11:
@CloudyPadmal  I completed fetching data in LA but I cannot make a PR unless @Avjeet completes the Wave Generator Instrument as my PR is partly dependent on it. Currently I was using the old Control Activity class but I can't make a PR using it. So I am working on bugs in other instruments.

CloudyPadmal, 2018-06-11:
If a specific feature is required by an instrument, please implement that in the instrument class. We cannot make dependable instruments. If someone makes a change to that other instrument, the dependable instrument will fail right? So add whatever the functionality you need in Logic Analyzer and open a PR. Later we can remove them if necessary.

Vikum94, 2018-06-11:
Wait for the PRs https://github.com/fossasia/pslab-android/pull/941 https://github.com/fossasia/pslab-android/pull/949

cweitat, 2018-07-13:
Fixed on PR #1216

closed 2018-07-13 by cweitat