fossasia/pslab-app #1031: Flow of sensors

cweitat, 2018-06-13:
**Actual Behaviour**

User have to press on phone back button to go back to previous screen 

**Expected Behaviour**

There should be a hamburger icon to let user to go back to Instruments

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

GSoC or interns

Avjeet, 2018-06-13:
I will work on this issue.

Avjeet, 2018-06-14:
@cweitat some activites dont have the app bar so how could I put this icon in them?

cweitat, 2018-06-14:
@Avjeet I understand, so please discuss with @CloudyPadmal and the rest to see what is the best way to implement this; either to all or some activities.

Avjeet, 2018-06-14:
@cweitat 👍 ok will do

Avjeet, 2018-07-11:
@cweitat I have added the hamburger icon in activities which have app bar but there is one problem as I am thinking of adding those list of instruments in navigation panel as shown in image below. So should I include all the instrument or just those in which hamburger icon is there.
[image]

cweitat, 2018-07-15:
Leave this open for now until we find a way to a way to access the menu from every instrument

Smit2010, 2019-03-15:
I would like to work on this. Can I take this issue?

Smit2010, 2019-03-15:
The basic idea is to open the navigation drawer from every instrument activity right?
If yes, then I would like to work on this issue. Can I take this up?
@cweitat @CloudyPadmal

Smit2010, 2019-03-17:
What exactly are you expecting with this issue? Implementing a back button to navigate back to instrument activity or implementing a navigation menu same as in MainActivity?
@cweitat @Avjeet @CloudyPadmal

mariobehling, 2019-03-29:
In the meantime the implementation in most instruments was done differently with an back arrow. Therefore everyone please follow the same implementation in all instruments. Thanks.

closed 2019-03-29 by mariobehling