fossasia/pslab-app #1033: No I/O takes place in Logic Analyzer instrument

harsh-2711, 2018-06-14:
**Actual Behaviour**

Only placeholder data is used to plot the graph 

**Expected Behaviour**

There should be plot of the data that is generated by user

**Steps to reproduce it**

Observe Logic Analyzer instrument

**Would you like to work on the issue?**

Yes

harsh-2711, 2018-06-14:
@CloudyPadmal I started fetching data from Logic Analyzer and I used (a * 0xff) | ((b * 0xff) << 8) to combine two bytes to form an integer. Is this correct ? The data plotted after modifying the fetched data is as shown

[la_final.zip](https://github.com/fossasia/pslab-android/files/2100407/la_final.zip)

CloudyPadmal, 2018-06-14:
@harsh-2711 yes how you combine two bytes is correct. I cannot view the video as it shows as no read or write access.

harsh-2711, 2018-06-14:
@CloudyPadmal  Go to properties -> permissions -> Give both permissions. Same problem existed for me but it is default by GitHub I guess

CloudyPadmal, 2018-06-14:
@harsh-2711 this looks nice! Can you make a PR for this one?

PR #1044 (2018-06-18): feat: Implemented One Channel Mode Logic Analyzer
Fixes #1033 

**Changes**: 
Implemented Single Channel Mode Logic Analyzer for ID1 pin

**Screenshot/s for the changes**: 
[la_one_channel.zip](https://github.com/fossasia/pslab-android/files/2104065/la_one_channel.zip)
For Linux: Right click on extracted file -> Properties -> Permissions -> Give read permission to see the video

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifica [... cut]
closed 2018-06-18 by CloudyPadmal