fossasia/pslab-app #1081: I2C read issues

Vikum94, 2018-06-19:
**Actual Behaviour**

When using mpu6050 sensor with I2C, the data is off the chart and does not get updated. 

**Expected Behaviour**

Should receive accurate data continuously 

**Steps to reproduce it**

Connect sensor MPU6050 to I2C pins -> Go to Sensors -> MPU6050 -> Click the play button
Observe that the values are not updating. 

**LogCat for the issue**

06-19 08:31:58.023 14942-14997/org.fossasia.pslab E/PacketHandler: Error in packetHandler Reading
06-19 08:31:58.023 14942-14997/org.fossasia.pslab V/CommunicationHandler: TO read : 1
06-19 08:31:58.073 14942-14997/org.fossasia.pslab E/CommunicationHandler: Read Error: 1
06-19 08:31:58.218 14942-14976/org.fossasia.pslab V/CommunicationHandler: TO read : 14
06-19 08:31:58.218 14942-14976/org.fossasia.pslab E/CommunicationHandler: Read Error: 13
06-19 08:31:58.218 14942-14976/org.fossasia.pslab E/PacketHandler: Error in packetHandler Reading
06-19 08:31:58.218 14942-14976/org.fossasia.pslab V/CommunicationHandler: TO read : 1
06-19 08:31:58.273 14942-14976/org.fossasia.pslab E/CommunicationHandler: Read Error: 1
06-19 08:31:58.418 14942-14994/org.fossasia.pslab V/CommunicationHandler: TO read : 14
06-19 08:31:58.423 14942-14994/org.fossasia.pslab E/CommunicationHandler: Read Error: 13
06-19 08:31:58.423 14942-14994/org.fossasia.pslab E/PacketHandler: Error in packetHandler Reading
06-19 08:31:58.423 14942-14994/org.fossasia.pslab V/CommunicationHandler: TO read : 1
06-19 08:31:58.478 14942-14994/org.fossasia.pslab E/CommunicationHandler: Read Error: 1

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**

Yes

cweitat, 2018-07-11:
Progress update?

cweitat, 2018-07-16:
@harsh-2711 @Avjeet @abhinavraj23 @CloudyPadmal due to unavailability of Vikum, who can takeover this issue?

cweitat, 2018-08-07:
@CloudyPadmal is this fixed?

CloudyPadmal, 2018-08-08:
Hello @cweitat ! Currently me and @abhinavraj23 are working on a solution to fix this issue!

marcnause, 2025-01-13:
Can this issue be closed? I2C works fine for newly implemented sensors and old sensors will be checked in https://github.com/fossasia/pslab-android/issues/2594,

CloudyPadmal, 2025-01-16:
Fixed in #2527

closed 2025-01-16 by CloudyPadmal