fossasia/pslab-app #1127: New Logo For PSLab (Proposal)

ahyar92, 2018-06-23:
Hi sir, I am Ahyar,
Greetings :)

I would like to propose a new logo for your application. I created this logo to give a new impression to your application. you can see my design result below :). If you like it, just reply to this :)
Thanks

[image]

Avjeet, 2018-06-24:
@ahyar92 icon design is good, but I think the icon should be more directed towards electrical component and not a flask as in your icon because PSLab is open source project related to electronic components.

cweitat, 2018-07-09:
@ahyar92 any reasons why the logos are not the current red we are having or why we should not retain the current red?