fossasia/pslab-app #1141: Voltage measuring issues at SEN , CAP and AN8 pins

abhinavraj23, 2018-06-26:
**Actual Behaviour**

Voltage measurement at SEN CAP and AN8 shows wrong value in multimeter.

**Expected Behaviour**

Voltage measurement should be accurate.

**Steps to reproduce it**

Change in the science lab codes

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

Not applicable

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-06-27:
@abhinavraj23 please note that SEN and CAP pins are not used to measure voltages. SEN pin will show 3.3V all the time if no resistor is connected. Only AN8 pin should take into consideration.

abhinavraj23, 2018-06-27:
@CloudyPadmal But still voltage across them can be measured using the getVoltage just there is some bugs in it which I will fix :+1:

CloudyPadmal, 2018-06-27:
OK let's see your fix first. FYI we measure resistance using a voltage divider technique by measuring voltage at SEN pin internally. We don't recommend using SEN or CAP pins to measure voltage of an external source.

abhinavraj23, 2018-06-27:
@CloudyPadmal I was performing the experiment where I connected the PV1 pin to the SEN pin and using the multimeter I tried to measure the voltage, so it was working fine except for a bug of getting multiplied by a factor of two.

CloudyPadmal, 2018-06-27:
I kinda suspect when you change those settings, it will affect resistance measurement. I'm not sure. Let's see.

abhinavraj23, 2018-06-27:
@CloudyPadmal I am constantly doing work on the capacitance issue but that is just not happening, no matter how much I try, so I also need other tasks to work because I can't just spend all my time on just one issue, which I have tried a lot to solve but still have problems.

abhinavraj23, 2018-06-27:
@CloudyPadmal Yes I can continue my work on that issue and side by side work on other issues as well so that i can be more productive.If I just work on the capacitor issue which i don't even know how much time it will take then it will not make sense.

CloudyPadmal, 2018-06-27:
OK do whatever pleases you!

PR #1152 (2018-06-29): refactor: resolved voltage reading issues in AN8, SEN and CAP pins
Fixes #1141 

**Changes**: 
The voltages were getting multiplied by a factor of two which was fixed.

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does  [... cut]
closed 2018-06-29 by CloudyPadmal