fossasia/pslab-app #1163: Multimeter UI lines confusing

mariobehling, 2018-06-30:
The multimeter central UI lines (horizontal and vertical) of the device give an impression of a separation that does not reflect the device actually and could confuse the user.
[image]

Multimeters often have another UI, where the lines to indicate an area are separated and differently drawn. Please implement this similarly.

[image]

abhinavraj23, 2018-07-03:
[image]
@CloudyPadmal @harsh-2711 @Vikum94 @Avjeet Have a look at this UI , I was thinking we could implement a UI based on it

abhinavraj23, 2018-07-03:
If you guys feel any change is required or you have a better idea then please do comment :+1:

CloudyPadmal, 2018-07-05:
Hello @abhinavraj23 , what is the knob design you have in mind?

abhinavraj23, 2018-07-06:
@CloudyPadmal I have changed the knob design it is available in my PR #1115

closed 2018-07-15 by cweitat