fossasia/pslab-app #1165: implement the feature to produce the waves using phone audio jack

Avjeet, 2018-06-30:
**Actual Behaviour**

Currently their is no such functionality available to produce different from audio jack of phone

**Expected Behaviour**

It is an additional functionality,  in this, the audio jack of the phone can be used to produce different waves like sine, square, triangular which can be analyzed in the oscilloscope. 

**Steps to reproduce it**

Wave generator -> Implement the functionality using AudioTrack class provided by android 

**Would you like to work on the issue?**

Yes

harsh-2711, 2018-06-30:
@Avjeet It would be great if you first focus on the digital waves generator pins as it's difficult for me to everytime use `Control Activity` for testing and then remove the changes before commit. No digital pins i.e. SQR1, SQR2, SQR3, and SQR4 are working correctly.

ho-dor, 2019-06-27:
I would like to work on this. 
Reference :- [https://blog.fossasia.org/filling-audio-buffer-to-generate-waves-in-the-pslab-android-app/](url)

ho-dor, 2019-07-02:
@neel1998 @CloudyPadmal @mariobehling I don't think this is any different from the in-built mic functionality that we have already as we are using AudioJack instance for that functionality as well. This can be closed.

neel1998, 2019-07-03:
@CloudyPadmal @mariobehling  please close this

neel1998, 2019-07-10:
@CloudyPadmal  @mariobehling can we close this?

CloudyPadmal, 2019-07-10:
This is about producing waves; not observing. Do we have that feature?

neel1998, 2019-07-10:
I didn't understand your point. Currently we are displaying the wave produced by the sound on the oscilloscope. What do you mean by producing...??

CloudyPadmal, 2019-07-10:
Can you read and understand the **title** of this issue? You can also read the issue body if it's still hard for you to understand.

neel1998, 2019-07-11:
@CloudyPadmal here is what i hve understood, the user will decide some wave let's say some sine/triangular wave with freq,amp and phase, and we are suppose to produce a sound wave based on that. For example, whistling is a kind of a sine wave, so if user gives sine wave to this feature, a sound like whistle should be produced. Am i correct?

CloudyPadmal, 2019-07-11:
https://blog.fossasia.org/filling-audio-buffer-to-generate-waves-in-the-pslab-android-app/
https://blog.fossasia.org/using-the-audio-jack-to-make-an-oscilloscope-in-the-pslab-android-app/