fossasia/pslab-app #1166: Size of Carousel Picker not proper for smaller screens

harsh-2711, 2018-06-30:
**Actual Behaviour**

The size of the Carousel Picker in Logic Analyzer instrument decreases very much on small screen mobiles

**Expected Behaviour**

The size should remain uniform on all screens

**Steps to reproduce it**

Go to Logic Analyzer -> Observe Carousel Picker on different screen size mobiles

**LogCat for the issue**

Not applicable

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-07-01:
@harsh-2711 I have a suggestion. What if we remove the "SELECT" option from that and have only the numerical values with 1 channel as default? Is it difficult to implement?

harsh-2711, 2018-07-01:
No. It's not. @CloudyPadmal I will remove it.

CloudyPadmal, 2018-07-01:
@harsh-2711 great!

PR #1169 (2018-07-09): fix: Solved the bugs in Carousel Picker
Fixes #1166 

**Changes**: Solved the bugs coming in setting the Carousel Picker's size for different screens and also removed the hint text option from the Carousel

**Screenshot/s for the changes**: 
6 inch screen 
[image]

5.5 inch screen
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I [... cut]
closed 2018-07-09 by CloudyPadmal