fossasia/pslab-app #1201: Adding FAQs to the Help and Feedback section

yatri1609, 2018-07-09:
**Actual Behaviour**
Currently Help and Feedback section doesn't have anything.

**Expected Behaviour**
We can include some FAQs and our official e-mail id so that of the doubts are not solved through those FAQs one can contact through it.

**Steps to reproduce it**
Click on Help and Feedback from the side menu

**LogCat for the issue**
N/A

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

yatri1609, 2018-07-09:
@CloudyPadmal Can I work on this issue?

cweitat, 2018-07-09:
Before implementing, find out the following:
1. Check and finalise what to include in the list of FAQs
2. Have a template for FAQs and answers
3. Structure and flow of the help section (e.g. simple few questions with accordion?)

CloudyPadmal, 2018-07-10:
Yes you can @yatri1609 . Before getting started, please create a Google Doc and collect the data as @cweitat mentioned and share it in Gitter channel.

yatri1609, 2018-07-10:
Surely @cweitat @CloudyPadmal

cweitat, 2018-07-11:
#1213

cweitat, 2018-07-11:
#1213

cweitat, 2018-07-11:
fossasia/pslab-android#1213

CloudyPadmal, 2018-07-14:
@yatri1609 I have added a comment. @abhinavraj23 @Avjeet @harsh-2711 @Vikum94 please review the doc and add your input

yatri1609, 2018-07-16:
Yes @cweitat I'll start working on the layout

PR #1243 (2018-07-21): feat: Added FAQs section instead of Help and Feedback
Fixes #1201 and #1213 

**Changes**: 
Changed the Help and Feedback section to FAQs section and added the FAQs as discussed

**Screenshot/s for the changes**: 

|  |  |  |
| ----------- | ----------- | ---------- |
|[image]|[image]|[image] |

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [... cut]
closed 2018-07-21 by CloudyPadmal