fossasia/pslab-app #1202: csv file not being created if permission is not given expilicitly

Avjeet, 2018-07-09:
**Actual Behaviour**

For data loggingthe sensor activities are using csv looger which requires storage permission explicity if the storage permission is not there the file doe not get created. 

**Expected Behaviour**

Add permission dialog box to check if permission is given to the app for storing CSV file and if not request permission for the same from the app only.

**Steps to reproduce it**

CSVLogger-> check for persmission

**Would you like to work on the issue?**

Yes

woshikie, 2018-07-11:
Does user have access to the CSV file? Else can we make do without writing to storage?

Avjeet, 2018-07-11:
if we explicitly provide the permission for storage then only the file is being created else the file is not getting created

Avjeet, 2018-07-11:
I am including the request for permission inside the app itself

woshikie, 2018-07-11:
does the user have a choice of selecting if a "dump" file should be created?

PR #1212 (2018-07-21): fix: added permission check for storage in lux meter
Fixes #1202 

**Changes**: 
Added check for permission when the save button is clicked, if permission is not given it will prompt the user to give the permission.

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in ever [... cut]
closed 2018-07-21 by CloudyPadmal