fossasia/pslab-app #1203: Oscilloscope displaying incorrect readings

woshikie, 2018-07-09:
**Actual Behaviour**

Readings on the Oscilloscope is displaying incorrect readings

**Expected Behaviour**

The correct readings should be displayed 
- V+ = ~8v
- VR+ = ~6v
- V- = ~-8v
- VR- = ~-6v)

**Steps to reproduce it with Screenshots**

- Connect V+ to CH1
[image]
- Connect VR+ to CH1
[image]
- Connect V- to CH1
[image]
- Connect VR- to CH1
[image]

cweitat, 2018-07-10:
@Vikum94 please fix this asap

woshikie, 2018-07-11:
@Vikum94 can we have a status report?

PR #1221 (2018-07-13): Solved bug in voltage measurement in Oscilloscope instrument
Fixes #1203 

**Changes**: Doubled the output shown by the oscilloscope to match the exact input voltage

**Screenshot/s for the changes**: 

V+
[image]

V-
[image]

VR+
[image]

VR-
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file i [... cut]
closed 2018-07-13 by woshikie