fossasia/pslab-app #1207: Implement Open Street Map to show logged data

CloudyPadmal, 2018-07-10:
**Actual Behaviour**

Users can log data from sensors but they cannot view them from a map

**Expected Behaviour**

Users may log data in different locations and a feature is needed to let them easily view the data sets filtered from locations.

**Steps to reproduce it**

> Not applicable

**LogCat for the issue**

> Not applicable

**Screenshots of the issue**

> Not applicable

**Would you like to work on the issue?**

Yes

woshikie, 2018-07-11:
What is the issue? I don't understand at all

Avjeet, 2018-07-14:
@CloudyPadmal Hey the user can only see it's own datapoints or all the datapoints that is fetched all over the globe by other users too?

CloudyPadmal, 2018-07-15:
@Avjeet for now we have to show only the ones user has created. Then we can move onto displaying implementing a global data base to save other user's data and show them in a proper way.

CloudyPadmal, 2018-08-05:
Resolved in #1211

closed 2018-08-05 by CloudyPadmal