fossasia/pslab-app #1213: Convert Help & Feedback section to FAQ

cweitat, 2018-07-11:
**Actual Behaviour**
Currently help and feedback section sits together

**Expected Behaviour**
Feedback section should be removed from this section and renamed to FAQs

**Would you like to work on the issue?**
@yatri1609

cweitat, 2018-07-11:
#1201

cweitat, 2018-07-11:
#1214

yatri1609, 2018-07-11:
Ya surely I would like to work on this ! Thanks @cweitat

cweitat, 2018-07-11:
@CloudyPadmal help me to assign this issue. I can't seem to assign it to her

CloudyPadmal, 2018-07-12:
@cweitat there seems to be an issue in assigning. @yatri1609 are you a member of https://github.com/fossasia ? When I try to add you to PSLab-Developers team, it says you are not a member of the organization. Can you please check and make a request if you are not to be a member?

yatri1609, 2018-07-12:
@CloudyPadmal Ya I'll make a request to be a member

cweitat, 2018-07-15:
@yatri1609 have you requested?

yatri1609, 2018-07-16:
@cweitat Actually I'm not getting the join option when I open the FOSSASIA account on GitHub.

closed 2018-08-08 by cweitat