fossasia/pslab-app #1227: only oscillscope open from wave generator

Avjeet, 2018-07-14:
**Actual Behaviour**

Currently only oscillscope opens from wave generator on clicking view button

**Expected Behaviour**

The user should be able to other instruments like logic analyzer or multimeter on pressing view button.

**Steps to reproduce it**

Ideas required for this.

**Would you like to work on the issue?**

Yes

Avjeet, 2018-07-14:
@CloudyPadmal @cweitat For selecting other instruments on pressing view button should I create a dialog box with the list of instruments or should I create carousel picker inside the button similar to one in the logic analyzer.

cweitat, 2018-07-15:
What's the purpose of implementing this? Does the wave generator affects the values from other instruments other than the oscilloscope? If not, what is the purpose of this?

If you are saying to implement this for #1031, I still can consider.

Avjeet, 2018-07-15:
@cweitat Yeah the need is that currently to produce waves from wave generator user need to press view which open only oscilloscope and sometimes user need to open logic analyzer or may be multimete so I added a simple dialog with carousel picker so the user can choose the instrument he /she wants to open.
Without it first the user has to exit the wave generator and then open some other instrument.

Avjeet, 2018-07-15:
@woshikie yes we can but Ithink there is no need as Power source as app bar which can have hamburger icon to open any instrument from the navigation panel.

CloudyPadmal, 2018-07-15:
Thinking about this, the usage of wave generator is not to generate waves and view it from logic analyzer or oscilloscope but to output it to a design or an experiment done by a user. If a user need to view the output from wave generator they can explicitly browse to those viewing instruments and see.

Avjeet, 2018-07-16:
@CloudyPadmal I know that and this Dialog box only prompts the user to open any other instrument if user just wants to proudce waves and not view it simply cancel the dialog box or press anywhere on the screen the dialog box will cancel but waves will still be produced

cweitat, 2018-07-16:
Does Wave Generator affects the value for multimeter and logic analyzer as well? If yes then okay we can keep the dialog, else remove the instruments from the dialog box that has no effect.

PR #1228 (2018-07-26): feat: added dialog with intent buttons to open other instrument
Fixes #1227 

**Changes**: Added dialog box with with two buttons to choose the different instrument to open from the wave generator

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in  [... cut]
closed 2018-07-26 by CloudyPadmal