fossasia/pslab-app #1237: Display toast if instruments/function for button is not available

cweitat, 2018-07-17:
**Actual Behaviour**
Currently on instruments screen when we press on Compass, nothing is showing up.
User may think that is either the app has problem or phone not working well

**Expected Behaviour**
There should be a toast to inform the user that the instrument is under development and display message 
"Still in development. Updates to come." or other more suitable message that you think is suitable.

**Would you like to work on the issue?**
@yatri1609 or @abhinavraj23

cweitat, 2018-07-17:
@CloudyPadmal was there anything done about this?

abhinavraj23, 2018-07-17:
Sure I will work on it @cweitat

harsh-2711, 2018-07-17:
@cweitat @abhinavraj23 No need to work on this, I have completed the UI and mobile sensor detection and will soon complete the sensor implementation.

cweitat, 2018-07-17:
@harsh-2711 okay thanks but also keep in mind that we should implement this if in the future there is similar situation

closed 2018-07-17 by cweitat