fossasia/pslab-app #1277: [Multimeter] Unable to Measure Resistance and Capacitance

woshikie, 2018-07-21:
**Actual Behaviour**

Measuring resistance always show `INFINITY`
Measuring Capacitance either crashes the activity or hangs the app.

**Expected Behaviour**

Measure resistance and capacitance without accurately and without crashing the app.

**Steps to reproduce it**

- Measure a resistor across SEN
- Measure a capacitor across CAP

abhinavraj23, 2018-07-21:
@woshikie The problem in this issue has also come due to merging of pr #1221 , it was working fine till then. But as far as the capacitor is concerned it is still working fine , I have tested it several times.

abhinavraj23, 2018-07-24:
@CloudyPadmal can I close this issue as this was solved in pr #1272