fossasia/pslab-app #1278: [Multimeter] Incorrect Voltage SEN

woshikie, 2018-07-21:
**Actual Behaviour**

Voltage across SEN always reads 3.30 V, unless connected to GND, which then reads 0v

**Expected Behaviour**

Correct voltage should be read

**Steps to reproduce it**

Measure voltage across SEN

**Screenshots of the issue**
[image]

abhinavraj23, 2018-07-21:
@woshikie I think that is the property of SEN pin, it should show 3.30 , can you please confirm on it @CloudyPadmal .

CloudyPadmal, 2018-07-22:
SEN pin is dedicated for resistance measurement and would not recommend to use it as a voltage measurement pin. It is pulled up to measure resistance across the pin and GND using a voltage divider method. Voltage at 3.3V is intended.

closed 2018-07-22 by CloudyPadmal