fossasia/pslab-app #1279: [Multimeter] Add readable Voltage range

woshikie, 2018-07-21:
**Actual Behaviour**

Users are not informed that only CH1 and CH2 can read a free range of voltage (from presumably ± ∞V).
CH3, SEN, CAP, AN8 can only read from 0V to 3.30V

**Expected Behaviour**

Users should be informed on the voltage range readable for all the pins

abhinavraj23, 2018-07-21:
@woshikie Shall I add this information in the bottomsheet ?

woshikie, 2018-07-21:
@abhinavraj23 do you think it will look nice?

abhinavraj23, 2018-07-21:
@woshikie Yes I think we can put the information about range in the bottom-sheet :+1: , what do you think ?

woshikie, 2018-07-22:
wouldn't there be too much information if you put it there?

cweitat, 2018-08-08:
@abhinavraj23 just add the information to the bottom sheet. preferably near the top so is viewable

PR #1655 (2019-03-29): Fix #1279 pins voltage range added to multimeter bottomsheet
Fixes #1279 

**Changes**: Voltage range added to multimeter bottom sheet

**Screenshot/s for the changes**:
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My  [... cut]
closed 2019-03-29 by mariobehling