fossasia/pslab-app #1304: lux is always running

Avjeet, 2018-07-29:
**Actual Behaviour**

Currenlty lux meter is always fetching data from sensors and keeps running continously

**Expected Behaviour**

The lux meter should have play pause functionality so that the user can start fetching the lux data whenever it wants.

**Steps to reproduce it**

LuxMeterFragmentData -> create functionality to start sensor fetching on the play button click

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-07-29:
May be we can combine this and #1303 ?

Avjeet, 2018-07-29:
@CloudyPadmal In this issue I created the functionality so that the user can control the working of lux meter (i.e., start plotting graph and meter) like phyfox.

PR #1305 (2018-08-11): feat: extended record button to start sensor data fetching and recording
Fixes #1304 

**Changes**: 

- Extended the record button to start the meter and graph plotting(i.e sensor data fetching) and also to record data.
- Added new record button

> To do  
Store data in realm instead of CSV file.(That will be done in PR #1323 )

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications d [... cut]
closed 2018-08-11 by CloudyPadmal