fossasia/pslab-app #1308: Release PSLab Android app in Fdroid

CloudyPadmal, 2018-07-29:
**Actual Behaviour**

Package name changes to `pslab.org` was merged. Future fdroid deployments would not be possible with the older package name. 

**Expected Behaviour**

Have to make a request to Fdroid community to remove the current app and submit a new merge request with new package name

**Steps to reproduce it**

> Not applicable

**LogCat for the issue**

> Not applicable

**Screenshots of the issue**

> Not applicable

**Would you like to work on the issue?**

Yes. Similar changes need to be done at Google Play. 
CC: @mariobehling

mariobehling, 2018-08-23:
Ok, Playstore is done. What about Fdroid? If not possible with older name, please submit new one.

CloudyPadmal, 2018-10-21:
New app is released and the old one removed.

closed 2018-10-21 by CloudyPadmal