fossasia/pslab-app #1338: License confusion

IzzySoft, 2018-08-07:
**Actual Behaviour**

Two contradicting licenses are mentioned.

**Expected Behaviour**

A clear license.

**Steps to reproduce it**

Check the [LICENSE file](https://github.com/fossasia/pslab-android/blob/development/LICENSE) against the `README.md`

**LogCat for the issue**

n/a

**Screenshots of the issue**

n/a

**Would you like to work on the issue?**

I'd just like to know what license is the correct one: Apache-2.0 or GPL-3.0-only. While on it, and as background: Someone opened an [MR to include your app with F-Droid](https://gitlab.com/fdroid/fdroiddata/merge_requests/3531). That OK with you? I couldn't find a corresponding issue in your tracker here, so I thought to mention it.

CloudyPadmal, 2018-08-07:
Hi @IzzySoft ! Let me make the license change in the merge request in GitLab to Apache-2.0.

About the versions, we recently changed the whole package name from the old one. So I reset it to 1 and going to increment from there at each release. :)

IzzySoft, 2018-08-07:
Ah! I see, package name changed. OK, then my fears concerning `versionCode` are unwarranted :wink: Now if you just mark the two comments at Gitlab as "resolved", some of the devs can take over for merging.

I'll leave this issue here open as the Readme [still states GPLv3](https://github.com/fossasia/pslab-android/blob/development/README.md#license). I'd suggest you close it once that line was fixed. Thanks for your fast reaction!

CloudyPadmal, 2018-08-07:
Sure @IzzySoft ! Thank you very much for the review! :)

IzzySoft, 2018-08-07:
My pleasure! Thanks for sharing your app and contributing to F-Droid!

PR #1344 (2018-08-09): fix: license mismatch
Fixes #1338 

**Changes**: 
- Updated license to Apache 2.0
- Fixed broken link

**Screenshot/s for the changes**: 
> Not available

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [X] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [X] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [X] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [X] My code does not contain an [... cut]
closed 2018-08-09 by mariobehling