fossasia/pslab-app #1349: Removal of meters

cweitat, 2018-08-09:
**Actual Behaviour**
Currently in barometer and lux meter we are having two different design meters. Main aim of app is to display and plot data. The meter currently, other than providing a nicer design, has no use function.

Max Lux is shouldn't be capped at 10000. Refer to this doc: [Maximum Readings](https://www.noao.edu/education/QLTkit/ACTIVITY_Documents/Safety/LightLevels_outdoor+indoor.pdf)
This also applies to barometer. Every sensor also have different Min and Max so it will be tedious to identify the range for each.
 The meter design across the app is inconsistent is also as well. Removal of meters also unclutter the screen.

**Expected Behaviour**
Removal of meter and display readings individually. A simple example text in bold such as "Current Reading: 13.1 Lux" at the top of the graph is prominent enough. 

**Screenshots of the issue**
[image]

cweitat, 2018-08-09:
Opened to suggestions on how to proceed

cweitat, 2018-08-09:
@Avjeet @abhinavraj23 @CloudyPadmal @harsh-2711 @yatri1609

yatri1609, 2018-08-09:
@cweitat If we remove the meter, the UI will be similar to that it was before. Meters were added just to enhance the UI of the app.

yatri1609, 2018-08-09:
@cweitat I think we should define a particular layout for all the sensors in common so that the app looks consistent and more appealing to the users.

cweitat, 2018-08-09:
@yatri1609 was thinking of bar gauge compared to circle guage

yatri1609, 2018-08-09:
@cweitat Maybe we can try that out, but the vertical bar gauge will take up more space I guess. What kind of bar gauge are you thinking of as I don't think I'm getting the same picture in my head as you have. If you have any picture can you share it?

cweitat, 2018-08-10:
@yatri1609 horizontal, not vertical. Something like the image below but slight design change to fit the app and also probably removal of intervals in the middle and one colour throughout the bar.
[image]

yatri1609, 2018-08-11:
Ohk I think we can do that @cweitat

TheShivamGambhir, 2018-08-18:
@cweitat Can we just display the readings on the tip of the needle or on circumference of barometer, this will retain both the functionalities of an attractive ui plus is analogous of folding the horizontal barometer to a circular one.

yatri1609, 2018-08-19:
@cweitat let's see what looks best but I guess gauge at the tool looks best

yatri1609, 2018-08-20:
@cweitat I was thinking that rather than adding a horizontal gauge why dont we make a uniform meter gauge for all the sensors either like that of Barometer or that of LuxMeter? That will make the app uniform and also keep the UI attractive.

yatri1609, 2018-08-21:
What do you all think about this ? @abhinavraj23 @Avjeet @harsh-2711 @CloudyPadmal

yatri1609, 2018-08-22:
@cweitat what do you suggest about making the meter gauge same for all the UI ?

cweitat, 2018-08-22:
@yatri1609 yes please change to the same meter gauge and remove the max number

yatri1609, 2018-08-22:
@cweitat I would go with the Barometer type gauge is it fine?

cweitat, 2018-08-22:
@yatri1609 yes but please remove the Min and Max because it varies between the devices

yatri1609, 2018-08-22:
@cweitat yes I'll do that

CloudyPadmal, 2018-11-16:
We have now standardized to use blue color gauge as the sensor meter

closed 2018-11-16 by CloudyPadmal