fossasia/pslab-app #1365: App crashing after changing fragments in Lux Meter

harsh-2711, 2018-08-12:
**Actual Behaviour**

App crashes when the record button is pressed after the fragment is changed from Data Plot to Config and again to Data Plot.

**Expected Behaviour**

The app shouldn't crash

**Steps to reproduce it**

Add steps to reproduce bugs or add information on the place where the feature should be implemented. Add links to a sample deployment or code.

**LogCat for the issue**

E/AndroidRuntime: FATAL EXCEPTION: main
                  Process: io.pslab, PID: 5786
                  java.lang.NullPointerException: Attempt to invoke virtual method 'void com.github.mikephil.charting.charts.LineChart.invalidate()' on a null object reference
                      at io.pslab.fragment.LuxMeterFragmentData.startSensorFetching(LuxMeterFragmentData.java:350)
                      at io.pslab.activity.LuxMeterActivity.onOptionsItemSelected(LuxMeterActivity.java:260)
                      at android.app.Activity.onMenuItemSelected(Activity.java:3477)
                      at android.support.v4.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:407)
                      at android.support.v7.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:195)
                      at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:108)
                      at android.support.v7.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:108)
                      at android.support.v7.app.ToolbarActionBar$2.onMenuItemClick(ToolbarActionBar.java:63)
                      at android.support.v7.widget.Toolbar$1.onMenuItemClick(Toolbar.java:203)
                      at android.support.v7.widget.ActionMenuView$MenuBuilderCallback.onMenuItemSelected(ActionMenuView.java:780)
                      at android.support.v7.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:822)
                      at android.support.v7.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:171)
                      at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:973)
                      at android.support.v7.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:963)
                      at android.support.v7.widget.ActionMenuView.invokeItem(ActionMenuView.java:624)
                      at android.support.v7.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:150)
                      at android.view.View.performClick(View.java:6367)
                      at android.view.View$PerformClick.run(View.java:25032)
                      at android.os.Handler.handleCallback(Handler.java:790)
                      at android.os.Handler.dispatchMessage(Handler.java:99)
                      at android.os.Looper.loop(Looper.java:164)
                      at android.app.ActivityThread.main(ActivityThread.java:6753)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:482)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-08-13:
@harsh-2711 @Avjeet @abhinavraj23 during the last meeting we were discussing about removing the bottom navigation bar from the instrument. Can you make the fix addressing that issue too?

Avjeet, 2018-08-13:
@cloudypadmal yeah I have done that by moving the option to lux meter settings I will make the Pr today

Avjeet, 2018-08-13:
@cloudypadmal yeah I have done that by moving the option to lux meter settings I will make the Pr today.

naman653, 2018-09-23:
This issue has been solved in https://github.com/fossasia/pslab-android/pull/1370.

closed 2018-10-01 by Avjeet