fossasia/pslab-app #1366: Samsung J7 devices are not supported

CloudyPadmal, 2018-08-13:
**Actual Behaviour**

Mr. Cheng reported that the app is shown as incompatible with J7 models and when I checked with Google Play profile it says J7s are partially supported. Digging more into the issue there could be a problem in Manifest requirements or package size.

**Expected Behaviour**

App should try to support as much as possible devices.

**Steps to reproduce it**

Install using a J7

**LogCat for the issue**

> Unsupported
This device model is not supported in your app's APK manifest and hence users of this device model cannot install your app. 

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

abhinavraj23, 2018-08-13:
@CloudyPadmal I have a samsung J7 mobile but it works fine!

CloudyPadmal, 2018-08-13:
Is it the supported version? There is no install button shown in other J7s!

cweitat, 2018-08-13:
There's many types of J7 and depending on country and the year it came out. Would be good if @CloudyPadmal know what is the variant.

ho-dor, 2018-11-13:
@CloudyPadmal,@abhinavraj23 , it is shown incompatible with my J8 2018 device as well

CloudyPadmal, 2018-12-21:
With the new release [v2.0.8](https://github.com/fossasia/pslab-android/releases/tag/v2.0.8), compatible devices expanded from 4794 to 6829 supporting J7 and multiple other devices.

[image]

closed 2018-12-21 by CloudyPadmal