fossasia/pslab-app #1373: [Multimeter] Readings are not shown continuously

woshikie, 2018-08-18:
**Actual Behaviour**

User has to tap on "read" to get readings

**Expected Behaviour**

Reading should be updated continuously

**Steps to reproduce it**

Enter multimeter

**Screenshots of the issue**

[image]

cweitat, 2018-08-20:
To implement continuous reading and a hold button like in the multimeter.
One way was to replace the read button with hold.

Smit2010, 2018-09-27:
@cweitat @woshikie @abhinavraj23  I would like to work on this issue.

abhinavraj23, 2018-09-27:
Yes @Smit2010 you can go forward with it. If you have any doubt or want to test the feature with the device then do let me know :+1:

neel1998, 2019-06-15:
@CloudyPadmal  i think this can be closed now

CloudyPadmal, 2019-06-15:
Fixed in #1767

closed 2019-06-15 by CloudyPadmal