fossasia/pslab-app #1376: App crash in multimeter

CloudyPadmal, 2018-08-18:
**Actual Behaviour**

According to crash reports in Google Play, there is a crash causing due to following reasons;
> Input dispatching timed out (Waiting to send non-key event because the touched window has not finished processing certain input events that were delivered to it over 500.0ms ago. Wait queue length: 10. Wait queue head age: 7224.2ms.)

**Expected Behaviour**

This kind of says that the click event was too quick for it to process. May be adding a lock once clicked on READ button until a fetch is complete would fix it. @abhinavraj23 any suggestions?

**Steps to reproduce it**

> Not available. Captured from crash logs

**LogCat for the issue**

```
  at android.hardware.usb.UsbDeviceConnection.native_bulk_request (Native method)
  at android.hardware.usb.UsbDeviceConnection.bulkTransfer (UsbDeviceConnection.java:250)
  at android.hardware.usb.UsbDeviceConnection.bulkTransfer (UsbDeviceConnection.java:232)
  at io.pslab.communication.CommunicationHandler.read (CommunicationHandler.java:132)
  at io.pslab.communication.PacketHandler.getVoltageSummation (PacketHandler.java:126)
  at io.pslab.communication.ScienceLab.getCapacitance (ScienceLab.java:2250)
  at io.pslab.communication.ScienceLab.getCapacitance (ScienceLab.java:2211)
  at io.pslab.activity.MultimeterActivity$4.onClick (MultimeterActivity.java:178)
  at android.view.View.performClick (View.java:6305)
  at android.view.View$PerformClick.run (View.java:24840)
```

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

@abhinavraj23 can you take this up?

abhinavraj23, 2018-08-18:
@CloudyPadmal There is no app crashing in my device, also is this the latest version?

abhinavraj23, 2018-08-18:
@CloudyPadmal I have tested the multimeter from the play-store version of app but there was no issue of crashing

abhinavraj23, 2018-08-18:
@CloudyPadmal I have found the issue, it is in the capacitance measuring function, it is due to the infinite do while loop in the code, although the app will crash ONLY if there is any loose connection or the data fetching is not done properly, i will solve the issue :+1:

PR #1379 (2018-08-20): fix: solved issue of crashing in capacitance measurement
Fixes #1376 

**Changes**: 
There was an infinity loop in capacitance measurement, changed it to a finite loop, also handled null capacitance value in MultimeterActivity

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd> [... cut]
closed 2018-08-20 by CloudyPadmal