fossasia/pslab-app #1388: Wave Generator - Setting Sine Wave Channel to 3.1 kHz results in an output of 6.3 kHz

geekyjoyce72, 2018-08-30:
**Actual Behaviour**

When the frequency of the Sine Wave Channel is set to 3.1 kHz, the actual measured output frequency on an external scope and frequency counter is ~6.5 kHz (not constant)

**Expected Behaviour**

The frequency output should be exactly that of the user's setting.

**Steps to reproduce it**

1. Use waveform generator application
2. Set sine channel to a specific frequency
3. Click view button
4. Measure output of sine wave with external oscilloscope and frequency counter

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

Not applicable, refer to video in zip archive

[PSLab_WaveGen_Issue.zip](https://github.com/fossasia/pslab-android/files/2337368/PSLab_WaveGen_Issue.zip)

**Would you like to work on the issue?**

Unsure. Need time to understand PSLab more

cweitat, 2018-08-30:
@CloudyPadmal please have a look at this.

nsurampu, 2018-12-19:
If no one's working on this, I would like to go ahead and give this a try.

ho-dor, 2019-06-24:
@neel1998 I am working on this. Thanks.

ho-dor, 2019-06-25:
@neel1998 @CloudyPadmal @mariobehling @cweitat can you confirm this issue still exists ? On my device, it is working fine . 3.1 khz shows 3.1 khz wave

neel1998, 2019-06-25:
I don't have an actual oscilloscope so can't confirm the generated wave frequency

geekyjoyce72, 2019-06-25:
@ho-dor is this a app or firmware related thing? I'm not exactly sure how the app and firmware works at the moment, sorry

If anyone around (in Singapore that is) can pass me a device, I can measure the signal with my scope and frequency counter to veirfy

cweitat, 2019-06-25:
@geekyjoyce72 I think i passed you a device previously? Or I remembered wrongly.

geekyjoyce72, 2019-06-25:
@cweitat you had requested me to return you the device at the start of the year or end last year I think, so I have no devices on hand to test with

cweitat, 2019-06-26:
@geekyjoyce72 follow up with you on Gitter

geekyjoyce72, 2019-07-01:
Hi @ho-dor, I can confirm that this issue has been fixed.

Attached is the respective measurement on a digital oscilloscope
[image]

closed 2019-07-01 by geekyjoyce72