fossasia/pslab-app #1389: Oscilloscope - Signals distorted;

geekyjoyce72, 2018-09-01:
**Actual Behaviour**

Waveforms above 50 kHz appear distorted.

**Expected Behaviour**

Waveforms should appear as they are, as per specifications (there's no specification listed on the maximum frequency of the oscilloscope channels)

**Steps to reproduce it**

1. Apply signal using external function generator to oscilloscope channel of PSLab; verify with external oscilloscope
2. View signal in oscilloscope app; signal is distorted.
3. Measure output of buffer amplifier at signal conditioning stage. 

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

Refer to attached video @ https://gofile.io/?c=FZKiRT

**Would you like to work on the issue?**

Am willing to help.

geekyjoyce72, 2018-09-22:
Hi.

I've checked the video link, it seems to be working. The set-up is as such:

1) Set-up an external function generator to output a >=50 kHz square wave
2) Verify signals on PSLab oscilloscope
3) Measure and verify signals using external oscilloscope, after input op-amp.

Feel free to contact me if you need more assistance

neel1998, 2019-07-02:
@geekyjoyce72 can you please check whether this issue still exist with latest changes. Thanks

geekyjoyce72, 2019-07-02:
Hello,

For the sake of being meticulous with testing, this is the output measured in the application.

[image]

This is an issue that is **not** at all possible to fix in software. The reason is that this is an issue on the hardware design side of things, namely, the analog front end. 

This is the part of the circuit that I mentioned in the initial post:
[image]

And this is the output of the op-amp, measured on a digital oscilloscope:
[image]

The issue can be seen when the circuit is recreated and simulated in LTSpice:
[image]

The op-amp is filtering out the higher frequency components of the signal, and as such, the square wave looks distorted. As you can see as per the simulation, the "cut-off" frequency of the circuit is around 216 kHz. 

The components in that section of the circuit that are responsible for setting the "cut-off" frequency are R20 and C23. 

This is the basic circuit of a op-amp low pass filter: 
[image]

and the "cut-off" frequency is set by this formula: 1/2πCR1

I have filed this issue here as all the issues are being tracked here, as I am told. In the meantime, I am looking at redesigning and improving the overall hardware design.