fossasia/pslab-app #1392: New instrument; Power meter

cweitat, 2018-09-19:
**Actual Behaviour**
Currently there is no instrument that is able to measure watts.

**Expected Behaviour**
There should be an instrument to measure the watts
Can also integrate watts to multimeter, oscilloscope , etc

**Would you like to work on the issue?**

Anyone.

naman653, 2018-09-23:
I would like to work on the implementation of this feature. Don't know much about the codebase of this project but I'm willing to help and contribute.

ho-dor, 2019-01-06:
@cweitat i would like to work on this

riyasingh1004, 2019-02-21:
Hi, I would like to work on this issue.

neel1998, 2019-07-28:
@cweitat  should this be included in multimeter it self or there should be a separate screen ? May be we can combine #1426 and this instrument together in one screen.

cweitat, 2019-07-28:
@neel1998 this was mentioned few meetings ago to include within multimeter.

neel1998, 2019-07-28:
@cweitat  yes, but multimeter is already very crowded that's why I asked. If we are going to read watts for the same pins as there are currently in multimeter, may be we can add a separate block in display which shows watt as well.

neel1998, 2019-07-28:
@cweitat for which pins exactly we plan to read power?

cweitat, 2019-07-29:
Derive from voltage

neel1998, 2019-07-29:
@cweitat where can I find which resistor is connected to which pins on PSLab internally. So I can calculate the power using P = V^2/R. I looked into the PSLab data sheet but resistor information is not given there.

cweitat, 2019-07-29:
@CloudyPadmal

CloudyPadmal, 2019-07-29:
Well, I think there's no direct way to measure power. Will have to connect a known resistor to measure current and then measure the voltage drop across the equipment to calculate I^2R. Not sure about the practicality of this method though. :thinking:

CloudyPadmal, 2019-07-29:
Hardware based solution would be better. Moving this to pslab-hardware repo