fossasia/pslab-app #1398: Update splash screen

yatri1609, 2018-10-01:
**Actual Behaviour**
Currently there is no animation to the splash screen which doesn't look much attractive, and also add the hard-coded values to dimen file

**Expected Behaviour**
Add an animation to both logo and text of the splash screen and add hard-coded values to dimen.xml file

**Steps to reproduce it**
Open the app

**LogCat for the issue**
N/A

**Screenshots of the issue**

Where-ever possible add a screenshot of the issue.

**Would you like to work on the issue?**
Yes

hrprt96, 2018-10-10:
yes

anujji1999, 2018-10-24:
Hello,
I am a beginner here.
Can I work on this issue?

ho-dor, 2018-11-02:
can I work on this @yatri1609 ?

anujji1999, 2018-11-10:
Is anyone working on it?
Or shall i start?

ho-dor, 2018-11-10:
Sorry @anujji1999, I am working on it as nobody replied for 4 days.

anujji1999, 2018-11-10:
Ok

ho-dor, 2018-11-10:
I have solved this issue but I am having trouble in creating a PR on the development branch . When I am trying to create a PR, I am getting an error "There is nothing to compare.master and development branches are completely different commit histories".I forked the pslab-android repo and then made the change to it in my system and them pushed the changes to my forked repo master branch and now I am trying to create a PR to the development branch of fossasia repo. Can somebody please help @yatri1609, @wavicles @mariobehling @lorenzgerber @cweitat @woshikie @jithinbp.

imrajat7, 2018-11-10:
@ho-dor you should commit your changes in the development branch of the forked repository and then after that PR to the project's development branch.

ho-dor, 2018-11-10:
Thank You @imrajat7

PR #1438 (2018-12-01): Updated Splash Screen Animation
Fixes #1398 

**Changes**: Fade Animation added to the splash screen by changes in SplashscreenActivity

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications are done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines  [... cut]
closed 2018-12-01 by CloudyPadmal