fossasia/pslab-app #1415: Fourier Transform in Oscilloscope

CloudyPadmal, 2018-10-16:
**Actual Behaviour**

Fourier transform analysis in PSLab oscilloscope is not working

**Expected Behaviour**

Frequency analysis of input signals need to be displayed correctly in the Fourier Transform section in oscilloscop

**Steps to reproduce it**

Check out Fourier transform analysis in oscilloscope with a sine wave or any kind of signal fed into oscilloscope channels

**LogCat for the issue**

> Not applicable

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Yes

neel1998, 2019-04-02:
Can someone guide me how can I start working on this issue. Thanks

neel1998, 2019-05-11:
@CloudyPadmal  can you please elaborate on this issue and guide me how to go about it

CloudyPadmal, 2019-05-11:
Hi, first go through the source code related to FFT in PSLab python repo and Android repos. If you're not familiar with Fourier Transform, I suggest you do some research on what it is and how to calculate FFT from a signal. It'll be helpful. Once all that is done, we have to implement Fourier Transform in Oscilloscope instrument.

PR #1786 (2019-06-15): fft output of ch1 working fine
Fixes #1415 

**Changes**: Fourier Transform output generated for channel 1

**Screenshot/s for the changes**: 
[image]

[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not cont [... cut]
closed 2019-06-15 by CloudyPadmal