fossasia/pslab-app #1426: New instrument; Ammeter

cweitat, 2018-10-25:
Actual Behaviour
Currently there is no instrument that is able to measure ampere.

Expected Behaviour
There should be an instrument to measure the ampere

Would you like to work on the issue?

Anyone.

ajit2704, 2018-12-05:
@cweitat  I would like to work on it.

harsh-2711, 2019-01-08:
@ajit2704 Have you started working on it?

ho-dor, 2019-01-08:
Is anyone working on this?

cweitat, 2019-01-08:
@ho-dor you can work on this

ho-dor, 2019-01-13:
@cweitat my laptop has broken down . U can assign this issue to someone else if it is urgent . Thanks.

Juhi-Purswani, 2019-02-04:
Sir,
I would like to work on it,  if this issue has still not been solved. Do let me know.
Thanks

riyasingh1004, 2019-03-07:
I would like to work on this issue.

CloudyPadmal, 2019-06-17:
We can add a section named `mA` to existing multimeter to implement this feature without creating a new instrument. @neel1998 @ho-dor

neel1998, 2019-07-28:
@CloudyPadmal is there any way we can implement this functionality without some external sensor like acs712. I am not very sure if we can get Current information from PSLab directly that's why asking you. Thanks

CloudyPadmal, 2019-07-29:
Yes right. There's no direct way to measure.

neel1998, 2019-07-29:
So should I get an acs712 sensor or you suggest some other one?

CloudyPadmal, 2019-07-29:
We'll leave it for now. Probably a hardware solution will be the best. I'm moving this to pslab-hardware repo.