fossasia/pslab-app #1431: Issue with the multimeter and the accelerometer.

Circinusss, 2018-11-03:
In the multimeter section when we try to access the settings of the multimeter the app crashes unexpectedly. This happens frequently not just a few times and is noticed across devices.
Also, the dial of the multimeter is not responsive, it takes 3-4 touches on the dial to rotate the dial.

A similar problem is noticed in the case of the accelerometer. Even when we try to access the settings of the accelerometer the app crashes the same way as in the case of the multimeter.

Besides these issues, the app is responsive and fluent no other issues noticed.

Below are the screenshots of the issue faced.
[image]
[image]

Hope the developer fix these soon.🙂

imrajat7, 2018-11-04:
@Circinusss issues you mentioned above are already fixed.

cweitat, 2018-12-13:
Issue has been fixed in latest release.

closed 2018-12-13 by cweitat