fossasia/pslab-app #1436: Delete redundant experiment resources

CloudyPadmal, 2018-11-06:
**Actual Behaviour**

Application doesn't directly implement experiments similar to desktop app. All the resources are still in the source code

**Expected Behaviour**

These files need to be removed to make apk file size smaller.

**Steps to reproduce it**

> Not available

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Yes

Mirnal1409, 2018-11-06:
Can I work on this

CloudyPadmal, 2018-11-06:
Yes go ahead!

CloudyPadmal, 2018-11-16:
@Mirnal1409 How is the progress with this issue?

ananyaarun, 2018-12-31:
@CloudyPadmal , can i work on this issue ?

cweitat, 2019-01-02:
@ananyaarun yes

ho-dor, 2019-01-06:
@ananyaarun are you working on this ?

ananyaarun, 2019-01-06:
@ho-dor , presently no anyone can take this up.

harsh-2711, 2019-01-07:
@CloudyPadmal Can I start working on this issue? Its been up for long

CloudyPadmal, 2019-01-07:
Yes @harsh-2711 :+1:

PR #1524 (2019-01-09): refactor: Removed the unused experiments resources and files
Fixes #1436 

**Changes**: Removed the unnecessary experiments files and resources to decrease the size of the app

**Screenshot/s for the changes**: N/A

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [ ] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [ ] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+< [... cut]
closed 2019-01-09 by cweitat