fossasia/pslab-app #1452: PSLab Crash upon viewing logged Lux meter data

cweitat, 2018-12-05:
**Actual Behaviour**
App crash whenever try to open

**Expected Behaviour**
Show the other page with the recorded data.

**Steps to reproduce it**
Press record and quit the instrument. Return back to Lux meter and show the logged data
Upon clicking the chosen data set, it crashes.

**Screenshots of the issue**

[image]

CloudyPadmal, 2018-12-05:
Hi @cweitat ! I've tried this. Is this sequence right in the screen recording?
[LuxMeter.mp4.zip](https://github.com/fossasia/pslab-android/files/2649970/LuxMeter.mp4.zip)

cweitat, 2018-12-05:
@CloudyPadmal  The first sequence is correct. 1s to 12s.

cweitat, 2018-12-05:
I just send a feedback with the system log.

PR #1455 (2018-12-06): fix: crash in Luxmeter when no data is saved
Fixes #1452 

**Changes**: 
- When there were no recorded data as user closes the app quickly, app crashed. It was fixed and a message is displayed

**Screenshot/s for the changes**: 
> Not available

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included [... cut]
closed 2018-12-06 by CloudyPadmal