fossasia/pslab-app #1453: Recorded data of Lux meter does not equal to Playback

cweitat, 2018-12-05:
**Actual Behaviour**

Playback data is shorter than the recorded one

**Expected Behaviour**

Both should be the same

**Steps to reproduce it**

Add steps to reproduce bugs or add information on the place where the feature should be implemented. Add links to a sample deployment or code.

**Screenshots of the issue**

[image]
[image]

CloudyPadmal, 2018-12-05:
Recording is only done between the record button is pressed and the stop button is pressed. The graph view will keep on displaying sensor values.

cweitat, 2018-12-05:
the 2nd image, which is the screenshot, shows that it is still recording. The stop button was pressed right after the screenshot was made. So the time should even be longer than the 4.8s. 

Playback which is the 1st image is less than 4.8s

CloudyPadmal, 2018-12-05:
Record button was first pressed after 2s? Compare the slopes.

cweitat, 2018-12-05:
both are reflecting the correct data at the time. slopes is also correct. 3s to 3.6s is upslope.

cweitat, 2018-12-05:
@CloudyPadmal okay I got what you mean now. Thanks!

closed 2018-12-05 by cweitat