fossasia/pslab-app #1458: App Close when I use Power Source Tool

alvinindra, 2018-12-09:
**Actual Behaviour**
I get app close when i use this Power Source tool

**Expected Behaviour**
I hope the tool is repaired

**LogCat for the issue**
When i touch/click the power source tool
I Use Xiaomi Redmi 3s and Android Marshmellow

**Screenshots of the issue**
When i click/touch that close
But no display force close app, when i back is same problem
[image]

**Would you like to work on the issue?**
Sorry i can't, i just share the information of issue/bug :+1:

cweitat, 2018-12-11:
Can you send a crash log?

cweitat, 2018-12-11:
Duplicate of #1461

closed 2018-12-11 by cweitat