fossasia/pslab-app #1465: Crashes

ezvezdov, 2018-12-10:
**Actual Behaviour**

It crashes when I record data in the accelerometer and run power source.
Doesn't work graphics at "sensors" and multimeter (nothing happens when I run it)

**Expected Behaviour**
Multimeter should work. The application should not crash.

**Would you like to work on the issue?**

Yes
[image]

CloudyPadmal, 2018-12-13:
Hello @ho-dor ! Would you like to take this up?

CloudyPadmal, 2018-12-13:
The issue has some corrections to do.

Firstly, the views wouldn't respond if the PSLab device is not connected unless we're using inbuilt sensors such as in Lux Meter, Accelerometer etc. Multimeter requires the PSLab device to show any functionality.

Similarly all the external sensors need the PSLab device to interface them with the app.

Other issues are valid.

ho-dor, 2018-12-13:
@CloudyPadmal yes . I will work on this . Thanks.

CloudyPadmal, 2018-12-13:
Before you start, we have an abstract class for sensors `PSLabSensor` that contains all the templates needed to record and function. Please have a look at how luxmeter is designed and implement accelerometer in the same way. Thanks!

CloudyPadmal, 2018-12-17:
Hello @ho-dor  ! Any update on this?

ho-dor, 2018-12-17:
@CloudyPadmal I am having trouble with my internet connection. Will give you an update in a day or 2 . Thanks

CloudyPadmal, 2018-12-17:
Well, accelerometer instrument already has a guide and you can use the content from there.

ho-dor, 2018-12-18:
@CloudyPadmal A new menu will have to be created for Accelerometer similar to Lux meter. Also , there is no fragment for accelerometer like in the case of Lux meter which has LuxMeterDataFragment and a RealmObject like LuxData for accelerometer will be needed . Am I thinking right or wrong ? 0a

CloudyPadmal, 2018-12-18:
Yes, we need a menu, data fragment and a settings fragment along with layout file.

ho-dor, 2018-12-18:
@CloudyPadmal A new menu will have to be created for Accelerometer similar to Lux meter. Also , there is no fragment for accelerometer like in the case of Lux meter which has LuxMeterDataFragment and a RealmObject like LuxData for accelerometer will be needed . Am I thinking right or wrong ?

ho-dor, 2018-12-19:
The Data fragment layout will have to be created for 3 axes and So the currently used recycler view has to be removed ? @CloudyPadmal

CloudyPadmal, 2018-12-19:
Remove the scroll views and constraint the three graphs in the fixed view

ho-dor, 2018-12-25:
@CloudyPadmal I have solved the issue partially . But the problem is that I have to convert LuxDataFragment from one graph to 3 graphs for accelerometer. So , as of now x-axis is working fine but how to convert it for 3 axes ?

CloudyPadmal, 2018-12-26:
Only the `PSLabSensor` class is abstract. Other classes and fragments feel free to modify to have a working implementation. Send us a PR to review and we can progress from there.

ho-dor, 2019-01-02:
I have solved this issue . I will be creating a PR by tomorrow.

harsh-2711, 2019-01-08:
@ho-dor Any updates?

PR #1541 (2019-05-11): fix:Accelerometer Crash Fixed and Updated changes with current developmen…
Fixes #1465 

**Changes**: Accelerometer Data, Settings Fragment added and Adapter Removed

**Screenshots for the changes**:

<table>
<tr>
<td>[image]</td>
<td>[image]</td>
<td>[image]</td>
</tr>
</table>

**Checklist**: 

- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [ ] No modifications are done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd [... cut]
closed 2019-05-11 by CloudyPadmal