fossasia/pslab-app #1466: Unexpected Crash

KSNINJA, 2018-12-10:
**Actual Behaviour**
In multimeter , unexpected crash occurs on pressing delete after pausing the recording .
Please state here what is currently happening.

**Expected Behaviour**
The app should work smoothly.
State here what the feature should enable the user to do.

**Steps to reproduce it**
Go to multimeter , click on record , then pause and then on delete .
Add steps to reproduce bugs or add information on the place where the feature should be implemented. Add links to a sample deployment or code.

**Screenshots of the issue**
[image]

Where-ever possible add a screenshot of the issue.

**Would you like to work on the issue?**
No , just a review
Let us know if this issue should be assigned to you or tell us who you think could help to solve this issue.

ho-dor, 2018-12-26:
@CloudyPadmal @cweitat can I solve this in a similar way by using PSLabSensor abstract class ?

CloudyPadmal, 2018-12-26:
`PSLabSensor` class is only for sensors. Other instruments need to be handled individually.

ho-dor, 2018-12-26:
I will like to work on this then

abhinavraj23, 2019-01-04:
@CloudyPadmal @ho-dor I would like to take up the issue if it's okay

ho-dor, 2019-01-04:
@abhinavraj23 I want to learn the working of multimeter . I would like to help you in this if it's ok .

ho-dor, 2019-01-05:
Did this issue require the pslab device to solve ? @abhinavraj23

abhinavraj23, 2019-01-06:
No, I just added the condition that the recording will take place only when PSLab is connected which makes sense.

PR #1520 (2019-01-06): fix:solved the issue of multimeter crash while recording
Fixes #1466 

**Changes**:
Restricted data recording only while PSLab is connected

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines  [... cut]
closed 2019-01-06 by cweitat