fossasia/pslab-app #1517: Instrument doesn't works if decimal points is in commas (Localisation issue)

cweitat, 2019-01-02:
**Actual Behaviour**
Values are not shown properly if phone language is set to within Europe. 
The same issue happens in oscilloscope, power source, multimeter and others.

**Expected Behaviour**
Calculation of values should still happen regardless of the language set on the phone.

**Steps to reproduce it**
Set your phone default language to English (Germany) and open the app.

**Screenshots of the issue**
[image]

cweitat, 2019-01-07:
Solution could be detect the language/country of the phone settings and show the values respectively.

Yahhi, 2019-01-08:
I want to work on this issue

cweitat, 2019-01-08:
@Yahhi go ahead

Yahhi, 2019-01-10:
There are even places where the app crashes with something like this: 
FATAL EXCEPTION: main
    Process: io.pslab, PID: 14645
    java.lang.NumberFormatException: For input string: "-0,66"
        at sun.misc.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2043)
        at sun.misc.FloatingDecimal.parseDouble(FloatingDecimal.java:110)
        at java.lang.Double.parseDouble(Double.java:539)
        at java.lang.Double.valueOf(Double.java:503)
        at io.pslab.others.FloatSeekBar.getValue(FloatSeekBar.java:42)
        at io.pslab.fragment.TimebaseTriggerFragment$2.onProgressChanged(TimebaseTriggerFragment.java:144)
...
The main reason for all this behavior is the process of digit limitation (2 digits after the dot). DecimalFormat and Double.valueOf on the result are used everywhere, It uses default locale values for creating strings, but can't get a value from a string with a comma. I suggest using a different approach to limit digits.