fossasia/pslab-app #1518: Unstable oscilloscope readings

cweitat, 2019-01-02:
**Actual Behaviour**
The oscilloscope reading are not stable. The readings jump around especially when the flashing frequency becomes high or if time base goes up.

**Expected Behaviour**
readings should be smooth and not stuck

**Steps to reproduce it**
Do a simple experiment of flickering LED with wave generator

**Screenshots of the issue**
[image]

cweitat, 2019-01-07:
Similar to #1389

geekyjoyce72, 2019-02-28:
Why not I look into it? Do you have actual hardware samples on hand?

cweitat, 2019-06-22:
https://docs.google.com/document/d/1MJ8MnqumBS0ZbtPb2n0we-QWOodeTaVfiBo8jkOKfSQ/edit
try with this experiment but with the Android app

ho-dor, 2019-06-26:
@neel1998 I will have a go at this one. Thanks.

neel1998, 2019-06-26:
@ho-dor sure go ahead. But first make sure this issue still exists

ho-dor, 2019-06-26:
Yup. We also need to figure out whether #1388 exists or not.

cweitat, 2019-06-26:
@ho-dor try to do the experiment from the link I gave above. It's the closest one that I can find. Else you can just do what I mentioned in the issue above

ho-dor, 2019-06-27:
@cweitat I am unable to re-generate this issue with frequency above 4000 and maximum timebase possible. Can you try once this expt , I think this issue doesn't exist anymore maybe.

ho-dor, 2019-06-27:
Works perfectly for me oscilloscope even with max frequency and timebase

neel1998, 2019-06-27:
@ho-dor i also thought that after the new changes in the oscilloscope this issue might have been solved. But let's wait for confirmation from @cweitat

woshikie, 2019-06-27:
[image]
Still looks slightly glitchy, especially towards the end.
When reading 2 channels at the same time, the app glitches out

neel1998, 2019-07-06:
@woshikie can you please check with the latest app. The last gif you sent is not with the latest app version. Thanks

neel1998, 2019-07-18:
@cweitat  @woshikie  Have you been able to check this issue with the latest app?

neel1998, 2019-07-29:
@cweitat any updates on this?

woshikie, 2019-07-29:
[image]

sorry for the delay. It's still buggy

cweitat, 2019-08-06:
@neel1998 do you know the setup to acheive this readings?

neel1998, 2019-08-11:
@cweitat sorry for the late reply. I have been able to get this reading. I will try to get it fixed by today. :innocent:

PR #1923 (2019-08-15): fix: unstable oscilloscope bug fixed
Fixes #1518 

**Changes**: unstable oscilloscope bug fixed

**Screenshot/s for the changes**: 

**Before:
[image]

**After:
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does [... cut]
closed 2019-08-15 by CloudyPadmal