fossasia/pslab-app #1528: Suggestion: Improvement of Java Autoboxing and Unboxing

o0lwj0o, 2019-01-10:
Hi,
I have found some usage of “int a = Integer.valueOf(String)” in this project.
In fact the return type of “Integer.valueOf()” is “Integer” which is wrapper class.  Wrapper class stores in the heap rather than stack. It take more time to loopup. The other potential problem is that Java implement autoboxing and unboxing since JDK 1.5. If run this in low version JDK, it may get trouble.
Furthermore, in the above case, it need to cast Integer to int.
I recommend use “Integer.parseInt(String)” which return type is int to improve its performance.

PS:
This case also applies to “double-Doubel”, “float-Float”, “long-Long” and so on.

Detail websites and lines are listed below

Double.valueOf() 
| 73 91 | https://github.com/fossasia/pslab-android/blob/development/app/src/main/java/io/pslab/others/EditTextWidget.java
| 41 | https://github.com/fossasia/pslab-android/blob/development/app/src/main/java/io/pslab/others/FloatSeekBar.java

Float.valueOf() 
| 308 | https://github.com/fossasia/pslab-android/blob/development/app/src/main/java/io/pslab/fragment/ControlFragmentAdvanced.java

Integer.valueOf() 
| 616 641 743 768 869 894 | https://github.com/fossasia/pslab-android/blob/development/app/src/main/java/io/pslab/adapters/ControlMainAdapter.java

Best regards

harsh-2711, 2019-01-10:
@o0lwj0o Great catch. I will surely remove the autoboxing issue from all the classes. I also figured out that some of our lists also faces same issue. I will change them all at once.

Yahhi, 2019-01-10:
I already did it in https://github.com/fossasia/pslab-android/pull/1525

neel1998, 2019-04-03:
@harsh-2711  what's the status of this issue. If it not solved yet, i would like to take it up otherwise you can close this issue

closed 2019-04-29 by cweitat